pulquero/BatteryAggregator #77: Cell Voltage limitation - CVL

wetty, 2024-10-02:
Please include system details:

Hardware: 
- 2x Battery with 18x 280Ah connected in parallel to Multiplus II
- Venus on RPi 3B+
- 2x JK BMS connected to the Pi via dbus-serialbattery driver

Setup: DVCC
Everything is off except BatteryAggregator as BMS

I have two scenarios for reducing the charge voltage.

- Switch to float voltage: One of the batteries is ready to switch to float, which means the voltage is normally reduced. The second battery is still loading, so overall voltage needs to be the charge voltage. In this case BatteryAggregator only reduces the voltage to the float value when both batteries are ready for the float mode. So it keeps the highest Charge Voltage of the two batteries. This works absolutly correct in BatteryAggregator. 😃 
- Cell Voltage Limitation because overvoltage of one or more cells: SerialBattery has a CVL Function. This function prevents a bad balanced battery to overcharge the cell with the highest voltage and the bms to switch off because of overvoltage of this cell. In this case the overall charge voltage should be reduced to the minimum of the batteries. Here BatteryAggregator also stays at the highest voltage of the two batteries. Which I think is not correct.

Maybe it is a little bit tricky do decide which behaviour is correct. In my case when both batteries are in Bulk or Absorption mode then the minimum of the charging voltages is correct. When both batteries are in float mode, then the lowest voltage is correct. And when one batterie is in Bulk or Absorption and the other one is in float, then the voltage of the batterie in Bulk or Absorption should be correct.

wetty, 2024-10-03:
Setting for CVL was "min_when_balancing".
With this BatteryAggregator switches to the lowest value, which is in my example/Screenshoot the Foat/FloatTransition Value of Akku 1. 
Akku 2 is still balancing as the delta is 0.024V. So the overall Battery Charge Voltage should be the value of Akku 2. In my case 62.24V (18x 3.45 = 62.1V + 0.14V Drop Off). 
[image]

So I thing the CVL Setting/Calculation needs something that ignores the Float and Float Transition mode as long as another Battery is still in Bulk or Absorption Mode. Then it should take the lowest value of all the Buld/Absorption Voltages.

Only when all Batteries are in Float Or Float Transition Mode the highest of the Float Voltages should be used.

pulquero, 2024-10-04:
Can you check if the mode is available in dbus? This is the main issue with trying to implement CVL well, trying to extract the required information over dbus from the bms. Atm I am mainly dependent on /Io/AllowToCharge and /Balancing.

wetty, 2024-10-04:
The Mode as string value is available under the battery Info/ChargeMode
[image]
[image]

pulquero, 2024-10-13:
Does soc happen to be 100% or close to, when charge mode is float or float transition? If I can use soc then this will also work for people who don't use serialbattery.

pulquero, 2024-10-13:
I've also put a new version, 3.15, in 'main' for you to try. It should exclude any disconnected or floating batteries from the cvl calculation, see where that gets us.

wetty, 2024-10-17:
With this version the min_when_balancing seems not to work.
[image]

pulquero, 2024-10-19:
Try v3.16.

wetty, 2024-10-20:
Just installed 3.16 as the batteries were at 99% and one battery was already in float mode. In this case the float one should be ignored and the voltage of the one in bulk mode should be the correct one and not the float one.

[image]

pulquero, 2024-10-20:
what soc does the floating battery report as?

wetty, 2024-10-20:
> what soc does the floating battery report as?

As mentioned they were both on 99% 

[image]

Current setting is now max_when_balancing. But it also takes the lowest value

wetty, 2024-10-20:
Batteries again both in bullk mode

[image]

pulquero, 2024-10-20:
Ah, ok. v3.17 excludes batteries where /Info/ChargeMode (if available) contains "float".

wetty, 2024-10-21:
One Battery in Float is working. Hope I can get both batteries into float mode. Then they should be not exclude. 

[image]

wetty, 2024-10-23:
I currently did not get the second Akku fully loaded and balanced because of not enough sun. Final Test Result may take a few month until spring :-)

wetty, 2024-10-26:
Its not working. Seems that when the second Akku changes to float mode BatteryAggregator excludes both Akkus. 
But in this case both floating Akkus should be included again!

Both Akkus where in float mode (18x 3,375V)

[image]
[image]

pulquero, 2024-11-11:
I've added debug logging to track what's not working. You can turn on debug in config via https://github.com/pulquero/BatteryAggregator?tab=readme-ov-file#logging or live via dbus with /LogLevel.

wetty, 2025-03-07:
Finally the sun is back and V3.27 is working as expected.

The first battery ist in float and the second in absorption and the voltage in this case is the absorption voltage. Perfect. 
Thanks a lot. 

[image]

pulquero, 2025-03-08:
Great! I've tagged v3.27 to `latest` now.

closed 2025-03-08 by pulquero