pulquero/BatteryAggregator #83: Please help with the config

harald0815, 2024-12-25:
Hello, 

I need some help with my cofig.

Setup: 2BMS on a Smartshunt

BMS1: com.victronenergy.battery.ttyUSB0
BMS2: com.victronenergy.battery.ttyUSB1
Smartshunt: com.victronenergy.battery.ttyS7 n
[image]

what I want: 
current from "Smartshunt" ; 
voltages and SOC from BMS1;

I don't need BMS2

My config:

{
  "excludedServices": [
    "com.victronenergy.battery.ttyUSB1"
  ],
  "primaryServices": {
    "com.victronenergy.battery.ttyUSB0": [
      "/Soc"
    ],
    "com.victronenergy.battery.ttyS7": [
      "/DC/0/Current"
    ]
  },
  "capacity": 584
}

pulquero, 2024-12-25:
This looks fine on the face of it. What actually happens?

harald0815, 2024-12-25:
Hello, thanks for the quick answer. 

If I restart the cerbogx with this config, the battery aggregator does not start.

If I just exclude the smartshunt then it works

{
  "excludedServices": [
    "com.victronenergy.battery.ttyS7"
  ],
  "capacity": 584
}

pulquero, 2024-12-25:
Please check the log.

harald0815, 2024-12-25:
Config:

{
  "excludedServices": [
    "com.victronenergy.battery.ttyUSB1"
  ],
  "primaryServices": {
    "com.victronenergy.battery.ttyUSB0": [
      "/soc"
    ],
    "com.victronenergy.battery.ttyS7": [
      "/DC/0/Current"
    ]
  },
  "capacity": 584,
  "logLevel": "DEBUG"
}

Log:

root@Cerbo-GX-Lan:~# cat /var/log/BatteryAggregator/current
@40000000676c598a0742546c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyS7 /Dc/0/Current 3.0
@40000000676c598a0bc4524c INFO:main:Waiting for batteries (attempt 28 of 30)...
@40000000676c598a30cc64bc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.2
@40000000676c598a322f09cc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -10.66
@40000000676c598b0bcc800c INFO:main:Waiting for batteries (attempt 29 of 30)...
@40000000676c598b31eba7cc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.18
@40000000676c598b32c89664 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -9.6
@40000000676c598c06767cd4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyS7 /Dc/0/Power 153.23599243164062
@40000000676c598c06d77124 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyS7 /Dc/0/Current 2.9000000953674316
@40000000676c598c0bca41d4 INFO:main:Waiting for batteries (attempt 30 of 30)...
@40000000676c598c0bda2824 WARNING:main:No batteries discovered!
@40000000676c598c0bed69d4 INFO:main:Exit
@40000000676c599207c94364 INFO:main:Starting...
@40000000676c59920b116d6c INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
@40000000676c59920b881854 INFO:dbusmonitor:Found: com.victronenergy.settings, scanning and storing items
@40000000676c599315e8c99c INFO:dbusmonitor:       com.victronenergy.settings has device instance 0
@40000000676c59931764078c INFO:dbusmonitor:Found: com.victronenergy.system, scanning and storing items
@40000000676c599329073644 INFO:dbusmonitor:       com.victronenergy.system has device instance 0
@40000000676c59932b833e7c INFO:dbusmonitor:Found: com.victronenergy.battery.ttyS7, scanning and storing items
@40000000676c59932ea5101c INFO:dbusmonitor:       com.victronenergy.battery.ttyS7 has device instance 279
@40000000676c59933079bc94 INFO:dbusmonitor:Found: com.victronenergy.battery.ttyUSB0, scanning and storing items
@40000000676c59940008a4e4 INFO:dbusmonitor:       com.victronenergy.battery.ttyUSB0 has device instance 2
@40000000676c599401026a4c INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
@40000000676c5994086e979c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.09
@40000000676c599408802b9c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -4.59
@40000000676c599408902d44 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.08
@40000000676c599408b23434 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -4.13
@40000000676c599430a0ff5c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.07
@40000000676c59943179a064 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -3.72
@40000000676c5995068b2db4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyS7 /Dc/0/Power 137.38400268554688
@40000000676c599506e140dc DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyS7 /Dc/0/Current 2.5999999046325684
@40000000676c59950bca45bc INFO:main:Waiting for batteries (attempt 1 of 30)...
@40000000676c599530cedd8c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.06
@40000000676c599531683f2c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -3.35
@40000000676c59960bc7c904 INFO:main:Waiting for batteries (attempt 2 of 30)...
@40000000676c5996309ebd3c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -3.01
@40000000676c59970bcd1864 INFO:main:Waiting for batteries (attempt 3 of 30)...
@40000000676c5997327029ac DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.05
@40000000676c5997338d45f4 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -2.71
@40000000676c59980bcd147c INFO:main:Waiting for batteries (attempt 4 of 30)...
@40000000676c599830d8f77c DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -2.44
@40000000676c59990bc442ac INFO:main:Waiting for batteries (attempt 5 of 30)...
@40000000676c599930cee944 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Current -0.04
[... cut ...]
@40000000676c59d10bcf0494 INFO:main:Waiting for batteries (attempt 27 of 30)...
@40000000676c59d20bd44c24 INFO:main:Waiting for batteries (attempt 28 of 30)...
@40000000676c59d30bcebe44 INFO:main:Waiting for batteries (attempt 29 of 30)...
@40000000676c59d3342b3b74 DEBUG:com.victronenergy.battery.aggregator:Battery value changed: com.victronenergy.battery.ttyUSB0 /Dc/0/Power -0.0
@40000000676c59d40bf57854 INFO:main:Waiting for batteries (attempt 30 of 30)...
@40000000676c59d40bf5a34c WARNING:main:No batteries discovered!
@40000000676c59d40bf5af04 INFO:main:Exit

pulquero, 2024-12-27:
You need to either add "/DC/0/Voltage" or everything with []. Currently the minimum requirement for a battery is that it has a voltage. My suggested config would be:

```
{
"excludedServices": [
"com.victronenergy.battery.ttyUSB1"
],
"primaryServices": {
"com.victronenergy.battery.ttyS7": [
"/DC/0/Current"
],
"com.victronenergy.battery.ttyUSB0": []
},
"capacity": 584
}
```
That is, you want everything from the BMS, but you want the current from the shunt to override that of the BMS.