pulquero/BatteryAggregator #84: install failed unknown error 127 Raspberry Error after update.

BATMAR123, 2024-12-29:
Hallo, 
After a CDSystemAggregator update, the software hangs and cannot be installed or uninstalled. See pictures. Can someone please give me a command to uninstall it with PuTTY. 
Thanks

[image]

[image]

pulquero, 2024-12-29:
Do you have SetupHelper installed correctly?

BATMAR123, 2024-12-29:
[image]

I think so

pulquero, 2024-12-30:
You should be able to uninstall using setuphelper then. You could also try installing DCAggregatorService and BatteryAggregator again in case something didnt download properly.

catweazle0172, 2024-12-31:
I have the same issue here as well.
Seems to be that the new SetupHelper is not compatible anymore.
The location of the following directory is changed in SetupHelper: /data/SetupHelper/CommonResources to /data/SetupHelper/HelperResources/CommonResources
But after the change of the location other errors occurs.

pulquero, 2024-12-31:
I've pushed fixes to all my packages, try now.

BATMAR123, 2024-12-31:
Thanks for fixing. It works again.

closed 2024-12-31 by pulquero