pulquero/BatteryAggregator #85: Doc improvement request:

endurance1968, 2025-01-02:
What is the default CCL behaviour? 

1. "ir" 
2. "capacity" 
3. "count" 

or summing up the CCL given by the batteries connected? Which if the values are available by e.g. serialBattery would maybe most sense or?
Reading the documentation it is not clear to me.

Note: for CVL there is a remark which is the default behaviour.

thanks

closed 2025-01-03 by pulquero

pulquero, 2025-01-03:
Done. Default is "ir". Note, summing CCLs is unlikely to work well. If I have two batteries, batt A with 20A CCL, batt B with 80A CCL, then there is no guarantee if I charge with 100A that only 20A of that will go to batt A - it will depend on the internal resistances of the two batteries. For it to work, both batteries would need to know about the characteristics of the other.

endurance1968, 2025-01-06:
OK understand
How to debug the result of the calculations? Could not find anything in the logs (set to debug). maybe searching for the wrong keywords.

pulquero, 2025-01-06:
The emf and internal res

endurance1968, 2025-02-17:
Hi,

I am currently wondering how exactly the CCL and DCL is calculated.  I understand you cannot simply sumup the individual CCLs. (individual CCL of the packs is between 42 and 65A)

Below the values from dbus-spy. I was not able to understand how the voltage and resistance results into the CCL and DCL value. The internal resistance is in Ohm? I=U/R seems not really to fit if I take the lowest pair 51,495V/0,079Ohm=651,8A ?

How is it done?

```
Info/MaxChargeCurrent                                                                                                                                                                                   60.1145
Info/MaxChargeVoltage                                                                                                                                                                                      55.2
Info/MaxDischargeCurrent                                                                                                                                                                                64.6058
```

```
System/EMFs                                                                               [52.15716079660277, 51.49534366378724, 51.44588644983276, 51.925465848182746, 51.961214327020855, 51.501638545807644]
System/InternalResistances                                                    [0.037503603094174665, 0.07891162700898402, 0.041634812905311236, 0.017426379097962297, 0.003191500323036628, 0.0589016908306732]
```

pulquero, 2025-02-18:
You are sort of right, but you need to use the voltage drop over the internal resistance, not the voltage over the load U, ie. I = (EMF - U)/R. That is the total voltage potential EMF = U + IR, voltage drop across the load plus voltage drop across the internal resistance.

commit eac1122b (2025-01-03): Fixes #85. Add default CCL to docs.
commit d914fa32 (2025-01-08): Fixes #85. Add default CCL to docs.