pulquero/BatteryAggregator #87: Help on config for two JK BMS an Soc

leiweke, 2025-01-08:
Please include system details:

Hardware: RPi
2 x JKBMS
Setup: DVCC yes
Version 

Please provide output from dbus-spy.
[image]

[image]

Please attach any relevant logs. Log location is `/var/log/BatteryAggregator/current`.

Hello I have 2 JK BMS running with BatteryAggregator and it runs fine.
The SOC from JK BMS 1 is wrong and not correct and makes the SOC calculation in BatteryAggregator not correct.
The issue is clearly the SOC from JK BMS 1.

Would is be possible to use all data from JK BMS 1 and JK BMS 2 ,but only use the SOC from JK BMS 2?

Thanks Sebastian

closed 2025-01-08 by pulquero

pulquero, 2025-01-09:
I've made it possible in the latest version. You need to create a virtual battery that will get used instead of BMS 1. Then we can use the hook functionality to null out soc:
```
{
"virtualBatteries": {
"com.victronenergy.battery.batt1": ["class:hooks.FixedValueHook", "com.victronenergy.battery.ttyUSB2"]
},
"classes": {
   "hooks.FixedValueHook": {
      "path": "/Soc"
   }
}
}
```

leiweke, 2025-01-09:
Thanks a lot i will test it out

leiweke, 2025-01-09:
updated to version 3.25
It looks like it's working as SOC on Batt All is from Batt 2. but the config check give many errors:

`root@raspberrypi4:~# bash /data/BatteryAggregator/check_config.sh
Traceback (most recent call last):
  File "/data/BatteryAggregator/check_config.py", line 43, in validate_json
    validate(instance=json_data, schema=schema)
  File "/usr/lib/python3.8/site-packages/jsonschema/validators.py", line 1121, in validate
    raise error
jsonschema.exceptions.ValidationError: Additional properties are not allowed ('classes' was unexpected)

Failed validating 'additionalProperties' in schema:
    {'additionalProperties': False,
     'properties': {'auxiliaryServices': {'anyOf': [{'items': {'type': 'string'},
                                                     'type': 'array'},
                                                    {'additionalProperties': {'items': {'type': 'string'},
                                                                              'type': 'array'},
                                                     'type': 'object'}]},
                    'capacity': {'type': 'integer'},
                    'currentRatioMethod': {'enum': ['ir',
                                                    'capacity',
                                                    'count'],
                                           'type': 'string'},
                    'cvlMode': {'enum': ['max_when_balancing',
                                         'min_when_balancing',
                                         'max_always',
                                         'dvcc'],
                                'type': 'string'},
                    'excludedServices': {'items': {'type': 'string'},
                                         'type': 'array'},
                    'logLevel': {'enum': ['NOTSET',
                                          'DEBUG',
                                          'INFO',
                                          'WARNING',
                                          'ERROR',
                                          'CRITICAL'],
                                 'type': 'string'},
                    'primaryServices': {'anyOf': [{'items': {'type': 'string'},
                                                   'type': 'array'},
                                                  {'additionalProperties': {'items': {'type': 'string'},
                                                                            'type': 'array'},
                                                   'type': 'object'}]},
                    'virtualBatteries': {'additionalProperties': {'anyOf': [{'items': {'type': 'string'},
                                                                             'type': 'array'},
                                                                            {'additionalProperties': {'items': {'type': 'string'},
                                                                                                      'type': 'array'},
                                                                             'type': 'object'}]},
                                         'type': 'object'}},
     'type': 'object'}

On instance:
    {'classes': {'hooks.FixedValueHook': {'path': '/Soc'}},
     'virtualBatteries': {'com.victronenergy.battery.batt1': ['class:hooks.FixedValueHook',
                                                              'com.victronenergy.battery.ttyUSB2']}}

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/data/BatteryAggregator/check_config.py", line 80, in <module>
    is_valid, is_warning, message = check_json_file()
  File "/data/BatteryAggregator/check_config.py", line 76, in check_json_file
    return validate_json(data)
  File "/data/BatteryAggregator/check_config.py", line 51, in validate_json
    error_message += err.path[0]
IndexError: deque index out of range`

pulquero, 2025-01-09:
That's ok, schema is just out of date for the checker. I'll try to update it.

commit 0874ad09 (2025-01-08): Fixes #87. Add FixedValueHook.