pulquero/BatteryAggregator #88: Virtual Battery CLV

adcrafter27, 2025-01-24:
Hello, you’ve already helped me with my setup using Pylontech and JKBMS
https://github.com/pulquero/BatteryAggregator/issues/66

Would it now also be possible to specify the CVL for the virtual battery?

JKBMS CVL = 51.5
Pylontech CVL = 53.2
As a result, the virtual battery based on Pylontech also has 53.2.

Thanks to your support, I’ve successfully set up the Pylontech as a virtual battery with my own CCL, and it’s working perfectly. It would now be great if I could set the CVL of my virtual battery (Pylontech) to 51.5 so that all BMS have the same charge cutoff voltage.

Thanks to your support

pulquero, 2025-01-25:
Have a look at https://github.com/pulquero/BatteryAggregator/issues/87#issuecomment-2578917844. Here soc is overridden to be empty. You can do a similar thing with cvl, difference will be you also need to add "value": 51.5 in addition to the path because you require a non empty value.

adcrafter27, 2025-01-25:
Hello,

Thank you! I’ve looked into it and tested several things, but apparently, I’m too clueless to understand exactly how to add this to the configuration.

My current configuration looks like this without the CVL adjustments:

{
  "excludedServices": [
    "com.victronenergy.battery.ttyUSB"
  ],
  "virtualBatteries": {
     "com.victronenergy.battery.better_pylon": ["class:hooks.CustomChargingHook", "com.victronenergy.battery.socketcan_can1"]
  },

  "classes": {
     "hooks.CustomChargingHook": {
           "ccls": {
                  "46": 30,
                  "48": 60,
                  "51": 40,
                  "51.2": 10,
                  "51.5": 0
           }
      }
  },
 "currentRatioMethod": "capacity"
}

The maximum charge voltage is "cvl" or, as seen in dbus-spy, "Info/MaxChargeVoltage".

No matter what I tried, there was always some kind of error.

pulquero, 2025-01-25:
Try this

```
{
"excludedServices": [
"com.victronenergy.battery.ttyUSB"
],
"virtualBatteries": {
"com.victronenergy.battery.better_pylon": [
"class:hooks.FixedValueHook",
"class:hooks.CustomChargingHook", "com.victronenergy.battery.socketcan_can1"]
},
"classes": {
"hooks.CustomChargingHook": {
"ccls": {
"46": 30,
"48": 60,
"51": 40,
"51.2": 10,
"51.5": 0
}
},
   "hooks.FixedValueHook": {
      "path": "/Info/MaxChargeVoltage",
      "value": 51.5
   }
},
"currentRatioMethod": "capacity"
}

adcrafter27, 2025-01-25:
Thank you, it works, you are my savior.

Is there a way to donate something to you via PayPal, I would like to thank you for your great work

pulquero, 2025-01-26:
😀 Pleasure, glad it works. I really appreciate the gesture of a donation. I'm happy for you to "pay it forward" and donate to a cancer charity instead.