pulquero/BatteryAggregator #90: Charging limits of 2 Battery packs "unmerged" when packs have been cut off by BMS

Underground56, 2025-02-10:
Hi there,

I'm using Battery Aggregator to merge 2x 16S LFP 280AH batteries. The batteries are used in parallel and each pack uses a JK BMS, both of them connected to the victron system. The system was used with one battery pack for~2years, the second packs has been installed just now. Since both batteries are not the exact quality and capacity, the older pack is cut off by the BMS first when discharging. Basically this would be no problem, but here's something strange:

When adjusted the charging limits of the Battery aggregator, the charge limits of both packs (let's say 100A for each) are more or less added to each other, so I get around 200A charge limit by the battery aggregator.  This is what is expect and correct:

[image]

When the packs have been cut of by the BMSes and start charging again (e.g. the next morning), Battery aggregator does not add both charging limits anymore. In this case the lower charging limit of both packs (e.g. 100A) is used despite the fact that both batteries are connected again and charging:

[image]

Both pictures were taken the same time, with the only difference that I just "reinstall-local" command to initialize the charge limits.  

Can you help to solve this issue? If you need more data, feel free to contact me.

with best regards
Michael

tzapu, 2025-02-11:
i believe i noticed the same thing, got one battery limited at 120A and one at 190A and the limit after a while seems to max out at 120A

thanks for BA!

pulquero, 2025-02-11:
Config and log would be helpful. CCL adjustments are logged precisely to help address these issues.

Underground56, 2025-02-11:
Hope this is helpful:

*********************************************************config.ini file*******************************************************************
; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; Calculated for each Battery
MAX_BATTERY_CHARGE_CURRENT = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

GUI_PARAMETERS_SHOW_ADDITIONAL_INFO = True
CVCM_ENABLE = True

;---------------------------------------------------------------------------------------------------------
; Enable charge current control management based on temperature (True/False).
;---------------------------------------------------------------------------------------------------------
CCCM_T_ENABLE = True
; Enable discharge current control management based on temperature (True/False).
DCCM_T_ENABLE = True

; Set steps to reduce battery current.
; The current will be changed linearly between these steps if LINEAR_LIMITATION_ENABLE is set to True.
TEMPERATURES_WHILE_CHARGING      =    0,    2,    5,   10,   15,   20,   35,   40,   55
MAX_CHARGE_CURRENT_T_FRACTION    = 0.00, 0.10, 0.20, 0.40, 0.80, 1.00, 1.00, 0.40, 0.00

TEMPERATURES_WHILE_DISCHARGING   =  -20,    0,    5,   10,   15,   45,   55
MAX_DISCHARGE_CURRENT_T_FRACTION = 0.00, 0.20, 0.30, 0.50, 1.00, 1.00, 0.00

;--------------------------------------------------------------------------------------------------------
; Enable charge current control management based on SoC (True/False).
;--------------------------------------------------------------------------------------------------------
CCCM_SOC_ENABLE = True
; Enable discharge current control management based on SoC (True/False).
DCCM_SOC_ENABLE = True

; Set steps to reduce battery current.
; The current will be changed linearly between these steps if LINEAR_LIMITATION_ENABLE is set to True.
SOC_WHILE_CHARGING                 =   98,   95,   90,   85, 80
MAX_CHARGE_CURRENT_SOC_FRACTION    = 0.20, 0.30, 0.40, 0.70, 1.00

SOC_WHILE_DISCHARGING              =    5,   10,   15,   20
MAX_DISCHARGE_CURRENT_SOC_FRACTION = 0.20, 0.50, 0.90, 1.00

******************************Log taken before executing reinstall-local*************************************************************** 
root@raspberrypi4:~# cat /var/log/BatteryAggregator/current
@4000000067ab875e316bd13c INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.5942848413693425, 'ir'), (2.682694779317969, 'ir')]
@4000000067ab875e316bec94 INFO:com.victronenergy.battery.aggregator:DCL estimates: [306.10268954291377, 139.50012852453438]
@4000000067ab91710ab1e07c INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB2': 188.0, 'com.victronenergy.battery.ttyUSB3': 52.0}
@4000000067ab91710abbd35c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067ab91710ac57c04 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.5942848413693425, 'ir'), (2.682694779317969, 'ir')]
@4000000067ab91710acde074 INFO:com.victronenergy.battery.aggregator:DCL estimates: [299.7255501774364, 139.50012852453438]
@4000000067ab91ea234d123c INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB2': 188.0, 'com.victronenergy.battery.ttyUSB3': 40.0}
@4000000067ab91ea236be0f4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067ab91ea23a148d4 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.5942848413693425, 'ir'), (2.682694779317969, 'ir')]
@4000000067ab91ea23d5456c INFO:com.victronenergy.battery.aggregator:DCL estimates: [299.7255501774364, 107.30779117271877]
@4000000067ab929d23d2864c INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB2': 184.0, 'com.victronenergy.battery.ttyUSB3': 40.0}
@4000000067ab929d23edf9a4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067ab929d23f7f454 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.5942848413693425, 'ir'), (2.682694779317969, 'ir')]
@4000000067ab929d240951a4 INFO:com.victronenergy.battery.aggregator:DCL estimates: [293.348410811959, 107.30779117271877]
@4000000067ab93c90083dbdc INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB2': 180.0, 'com.victronenergy.battery.ttyUSB3': 40.0}
@4000000067ab93c90083fb1c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067ab93c900c9f144 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.5942848413693425, 'ir'), (2.682694779317969, 'ir')]
@4000000067ab93c900ca0c9c INFO:com.victronenergy.battery.aggregator:DCL estimates: [286.97127144648164, 107.30779117271877]
@4000000067ab94f51b293194 INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB2': 164.0, 'com.victronenergy.battery.ttyUSB3': 40.0}
@4000000067ab94f51b43c25c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
[... cut ...]
@4000000067aba52c2e187bd4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067aba52c2e1e30c4 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.99644128113879, 'capacity'), (2.0035714285714286, 'capacity')]
@4000000067aba52c2e22ebb4 INFO:com.victronenergy.battery.aggregator:DCL estimates: [151.72953736654804, 80.14285714285714]
@4000000067aba52f2a11013c INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.ttyUSB3 @ 51.23V is 0.01729356167124152+-0.0013346606204709115
root@raspberrypi4:~#

pulquero, 2025-02-13:
In the "before", internal resistance calculations are being used to determine CCL:
> @4000000067aba28d179e64f4 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.7104667401308071, 'ir'), (2.407525424505988, 'ir')]

In the "after", capacity is being used as a fallback to determine CCL as internal resistance requires a minimal number of measurements to be collected.
> @4000000067aba52c2e08d7ec INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.99644128113879, 'capacity'), (2.0035714285714286, 'capacity')]

this should switch back to 'ir' once enough measurements have been collected:
> @4000000067aba52f2a11013c INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.ttyUSB3 @ 51.23V is 0.01729356167124152+-0.0013346606204709115

tzapu, 2025-02-13:
i m seeing lots of logs for can0 internal resistance but little for can 1
my batteries do seem to have mismatched resistances, charging and discharging at different ratios, especially under high current, but when the low resistance one gets to 90%+ and the high resistance one starts ramping up current due to the voltage difference, the calculated IR and resulting CCL drops the charge current to nothing, 40A or so. was wondering if everything looks ok from your point of view, thanks

cheers

```
@4000000067add2bc26585284 INFO:com.victronenergy.battery.aggregator:Current ratios: [(31.887868662421333, 'ir'), (1.0323751700361448, 'ir')]
@4000000067add2bc26ea0364 INFO:com.victronenergy.battery.aggregator:CCL estimates: [3938.1517798090345, 41.295006801445794]
@4000000067add2bc27a22804 INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.socketcan_can0': 123.5, 'com.victronenergy.battery.socketcan_can1': 197.0}
@4000000067add2bc27bd58f4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.socketcan_can0', 'com.victronenergy.battery.socketcan_can1']
@4000000067add2bc27ec5834 INFO:com.victronenergy.battery.aggregator:Current ratios: [(31.887868662421333, 'ir'), (1.0323751700361448, 'ir')]
@4000000067add2bc2804be4c INFO:com.victronenergy.battery.aggregator:DCL estimates: [3938.1517798090345, 203.37790849712053]
@4000000067add2c8254d9174 INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.socketcan_can0 @ 51.380001068115234V is 0.006460109499570876+-0.0005066858141350008
@4000000067add2d2255d25bc INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.socketcan_can0 @ 51.400001525878906V is 0.006897525513039682+-0.0004717014941540323
@4000000067add2f1093575c4 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.socketcan_can0': 123.5, 'com.victronenergy.battery.socketcan_can1': 47.400001525878906}
@4000000067add2f10945c974 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.socketcan_can0', 'com.victronenergy.battery.socketcan_can1']
@4000000067add2f109588e24 INFO:com.victronenergy.battery.aggregator:Current ratios: [(34.86676696671778, 'ir'), (1.0295274716060951, 'ir')]
@4000000067add2f10a1e2814 INFO:com.victronenergy.battery.aggregator:CCL estimates: [4306.045720389646, 48.799603725063164]
@4000000067add2f30a7b0584 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.socketcan_can0': 123.5, 'com.victronenergy.battery.socketcan_can1': 43.70000076293945}
@4000000067add2f30a87b784 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.socketcan_can0', 'com.victronenergy.battery.socketcan_can1']
@4000000067add2f30a977aac INFO:com.victronenergy.battery.aggregator:Current ratios: [(34.86676696671778, 'ir'), (1.0295274716060951, 'ir')]
@4000000067add2f30aa4ad94 INFO:com.victronenergy.battery.aggregator:CCL estimates: [4306.045720389646, 44.990351294653486]
@4000000067add2f508fd23cc INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.socketcan_can0': 123.5, 'com.victronenergy.battery.socketcan_can1': 40.0}
@4000000067add2f5092ca3f4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.socketcan_can0', 'com.victronenergy.battery.socketcan_can1']
@4000000067add2f50940f32c INFO:com.victronenergy.battery.aggregator:Current ratios: [(34.86676696671778, 'ir'), (1.0295274716060951, 'ir')]
@4000000067add2f509526bd4 INFO:com.victronenergy.battery.aggregator:CCL estimates: [4306.045720389646, 41.18109886424381]
@4000000067add30e25d70cc4 INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.socketcan_can0 @ 51.13999938964844V is 0.005903119881608632+-0.00040754199386265475
@4000000067add30e26f5de8c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.socketcan_can0': 123.5, 'com.victronenergy.battery.socketcan_can1': 40.0}
@4000000067add30e273fbcb4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.socketcan_can0', 'com.victronenergy.battery.socketcan_can1']
@4000000067add30e27505e84 INFO:com.victronenergy.battery.aggregator:Current ratios: [(29.984247325956655, 'ir'), (1.0345014996854673, 'ir')]
@4000000067add30e275ed1bc INFO:com.victronenergy.battery.aggregator:CCL estimates: [3703.054544755647, 41.38005998741869]
@4000000067add30e27f83b2c INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.socketcan_can0': 123.5, 'com.victronenergy.battery.socketcan_can1': 197.0}
@4000000067add30e2812433c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.socketcan_can0', 'com.victronenergy.battery.socketcan_can1']
@4000000067add30e2826dcac INFO:com.victronenergy.battery.aggregator:Current ratios: [(29.984247325956655, 'ir'), (1.0345014996854673, 'ir')]
@4000000067add30e283d3754 INFO:com.victronenergy.battery.aggregator:DCL estimates: [3703.054544755647, 203.79679543803707]
@4000000067add31125433cec INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.socketcan_can0 @ 51.099998474121094V is 0.006321435464201386+-0.0003936027242838391
[... cut ...]
@4000000067add3cd2509ba44 INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.socketcan_can0 @ 50.619998931884766V is 0.01683188996540963+-0.0010280381047718034
@4000000067add3fb25bce95c INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.socketcan_can0 @ 50.599998474121094V is 0.017319209686387893+-0.0010888431731306794
@4000000067add40924d4a084 INFO:com.victronenergy.battery.aggregator:Internal resistance for com.victronenergy.battery.socketcan_can0 @ 50.619998931884766V is 0.017048061057263197+-0.0013018313466272103
```

tzapu, 2025-02-13:
yeah, it looks to work better with capacity for me, thank you!

Underground56, 2025-02-13:
> In the "before", internal resistance calculations are being used to determine CCL:
>
>
> In the "after", capacity is being used as a fallback to determine CCL as internal resistance requires a minimal [...]

Thank you so much. Unfortunately I think it never switches back to the "before" state and correct CCL calculation. The lower CCL does stay until I i execute "reinstall-local" to initialize the values.Then it's working again until the packs are cut-off by the BMS.

I just tried again to Charge with max. Power after both packs were fully discharged. It does not "add" the CCL values of both packs, it just takes the lower individual one:

[image]

@4000000067adfb4303980564 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfb43039824a4 INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfb430398382c INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 128.00001111980328]
@4000000067adfc6f1c4b7a14 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 144.0}
@4000000067adfc6f1c5987bc INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfc6f1c69a4bc INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfc6f1c703c3c INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 144.00001250977868]
@4000000067adfcab214a358c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 128.0}
@4000000067adfcab215e4644 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfcab21775454 INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfcab21824904 INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 128.00001111980328]
@4000000067adfce7274dc674 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 144.0}
@4000000067adfce72754988c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfce7275f27ac INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfce72763b3bc INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 144.00001250977868]
@4000000067adfd232d4bf85c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 128.0}
@4000000067adfd232d4e26f4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfd232d5650cc INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfd232d5c5bac INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 128.00001111980328]
@4000000067adfd9b3670e5b4 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 144.0}
@4000000067adfd9b3678bd84 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfd9b369b87c4 INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfd9b36a54bc4 INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 144.00001250977868]
@4000000067adfdd80049cc94 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 128.0}
@4000000067adfdd8005d9ae4 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfdd80061c934 INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfdd800777fcc INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 128.00001111980328]
@4000000067adfe1305b5d09c INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 144.0}
@4000000067adfe1305d42e0c INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067adfe1305f36e0c INFO:com.victronenergy.battery.aggregator:Current ratios: [(11510996.00888077, 'ir'), (1.000000086873463, 'ir')]
@4000000067adfe1305fce7d4 INFO:com.victronenergy.battery.aggregator:CCL estimates: [1657583425.2788308, 144.00001250977868]
@4000000067adfe4f09a517e4 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB2': 144.0, 'com.victronenergy.battery.ttyUSB3': 128.0}
[... cut ...]
@4000000067ae4fe32678230c INFO:com.victronenergy.battery.aggregator:Individual DCLs: {'com.victronenergy.battery.ttyUSB2': 88.0, 'com.victronenergy.battery.ttyUSB3': 40.0}
@4000000067ae4fe3268970bc INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB2', 'com.victronenergy.battery.ttyUSB3']
@4000000067ae4fe326898c14 INFO:com.victronenergy.battery.aggregator:Current ratios: [(14222386.863450883, 'ir'), (1.000000070311691, 'ir')]
@4000000067ae4fe326a2963c INFO:com.victronenergy.battery.aggregator:DCL estimates: [1251570043.9836776, 40.000002812467635]

pulquero, 2025-02-13:
It's only safe to add ccls if the batteries are identical. To have this behavior use the following config option:
	"currentRatioMethod": "count"

wetty, 2025-03-12:
Just my two cent to keep in mind:
The overall Charge Current Limit is not always the sum of CCL from each single Akku.
If one Akku is full an another is still loading, then it makes sense that the resulting CCL is the on of the akku that is still loading and not the sum of both.
I had some Issues with Charge Current Limit to high see Issue [https://github.com/pulquero/BatteryAggregator/issues/65](https://github.com/pulquero/BatteryAggregator/issues/65)

pulquero, 2025-03-12:
Yes, summing will only happen if the individual ccls are near identical too. In general, the total ccl is modulated by the individual ccls. Eg. if batt A has ccl of 10 and B has a ccl of 0, then based on batt A, total ccl of 2x10=20 is possible. For batt B, total ccl is 2x0=0. Therefore the selected total ccl will be 0, so that no individual ccl is exceeded.

closed 2025-04-25 by pulquero