pulquero/BatteryAggregator #92: SOC with 0% after reboot

Newti04, 2025-03-06:
Hi,
I use BatteryAggregator with 3x DalyBMS and dbus-serialbattery.
When i do a reboot of the System, BatteryAggregator shows for a short time, till all three dbus-serialbattery scripts are running and getting correct data, a SOC with 0%. 
This 0% is mostly send to VRM and in the VRM Battery SOC curve from this day you see the delta eq 0-80% 

Is it possible to wait, after a reboot, till correct data is collected to set the First SOC?

[image]

pulquero, 2025-03-08:
Could you check the voltage delta for me? It should wait until the voltage is correct. I need to ascertain if that is not working correctly or the soc is available some time after the voltage and thus I need to add an option for a further delay.

Newti04, 2025-03-08:
So, I rebooted my VenusOS and watched the starting process.
My Installation is VenusOS V3.54 large with this 2 lines in rc.local
bash /data/dbus-shelly-pro-3em-smartmeter/install.sh
bash /data/etc/dbus-serialbattery/reinstall-local.sh
The Package Manager with
SetupHelper v8.29
ShutDownMonitor v10.84
BatteryAggregator v3.26
dbus-pi v1.1

After starting the MPPTs the Shelly Pro 3EM started and then my 3 SerialBattery.
All SerialBattery shows the SOC 81% and Voltage 53,2V correct, no 0% SOC.
But BatteryAggregator does not start.
I saw in the PackageManager that there is a new version for BatteryAggregator and updated it to v3.27.
Immediately BatteryAggregator started now also with correct SOC and Voltage.

After the update, I tested a reboot 4 times and now the BatteryAggregator starts always after a SerialBattery has started.
No 0% SOC is shown on VenusOS and in VRM.
I am not 100% sure, but I think BatteryAggregator (with older Version v3.26 and earlier) started sometimes before a SerialBattery had started.

pulquero, 2025-03-08:
Ok, good. Feel free to reopen if it reoccurs.

closed 2025-03-08 by pulquero