pulquero/BatteryAggregator #93: No Float Transition with two Batteries

wetty, 2025-03-11:
Please include system details:

Hardware: RPi
2x 18S 280Ah 62,1V in Parallel
2x JKBMS
SerialBattery v2.0.20250207beta

Setup: DVCCOn
Shared Voltage and shared current sense enabled

Please provide output from dbus-spy. See Screenshot

[image]

Please attach any relevant logs. Log location is `/var/log/BatteryAggregator/current`.
See attached Log

[2025-03-11 Log.txt](https://github.com/user-attachments/files/19186328/2025-03-11.Log.txt)

One of the Batteries is already in Float Mode (60,75V). The other is still in Absorption Mode (62,10V). So the overall voltage is 62,1V, which ist totally fine.

When Akku 2 has finished his absorption Time he switches in Float Transition Mode. This normally reduces the voltage slowly from 62.1V to the 60.75V float voltage.

But here BatteryAggregator immediately switches to the lowest voltage of the two batteries, which is the float voltage of akku 1 and is 60,75V. Thats wrong! It should switch to the highest of the two voltages, which is the voltage of Akku 2, which is slowly reduced step by step from Absorption to float voltage.

The abrupt switch from 62.1V to 60.75V results in a big delta of the cell voltages, which then results that Akku 1 goes back to Bulk and then to absorption  (see screenshots). 

Feature of serialbattery v2, to define the cell difference when to switch back to bulk.

Expected behaviour:
Akku 1 Bulk  62,1V + Akku 2 Bulk  62,1V => Battery Bulk 62,1V
Akku 1 Float Transition  62,1V To 60,75V + Akku 2 Bulk 62,1V => Battery 62,1V (That ist working)
Akku 1 Float 60,75V + Akku 2 Bulk 62,1V => Battery 62,1V (That ist working)
Akku 1 Float 60,75V + Akku 2 Float Transition 62,1V to 60,75V => Battery 62,1V to 60,75V (Not Working, immediately switches to 60,75V)

Might be an idea to compare the actual Battery (2x akkus) with the Voltage to switch to. If there is a big difference, then make a slow transition.

[image]

[image]

[image]

wetty, 2025-03-12:
"cvlMode": "min_when_balancing"

It needs to be the min_when_balancing. When the OverVoltageProtection kicks in and reduces the Max Charge Voltage from one Akku, then this lower voltage has to be the voltage of the whole battery.
When I set it to max, then the OVP has no affect (not tested yet, but thats my understanding)

As the floating voltage is ignored [https://github.com/pulquero/BatteryAggregator/issues/77](https://github.com/pulquero/BatteryAggregator/issues/77) when one of the akkus is floating. 
It should be something like max_when_floating, when all akkus are floating, or in transition.

Apart from that, some kind of safety check, to prevent reducing the overall voltage in a big step might make sense. In my case the drop was nearly 1.5V

wetty, 2025-03-12:
max_when_balancing is not an option, as then OVP has no affect. Tested, see screenshot. Battery overall voltage should be 61.89V not 62.1ßV

[image]

pulquero, 2025-03-13:
I think the fix is to detect OVP because I can't think of any reason why it would make sense to ignore it. Is charge mode the only way to tell if OVP is active? I might have to look at the serialbattery source code.

wetty, 2025-03-13:
I think OVP has primarily nothing to do with it. Problem exist without any OVP
Akku1 is in  "Float,Linear Mode" and reached already the float voltage of 60.75V (But the battery overall has still the higher absorption or bulk voltage, as the second akku is still loading)
Akku2 is in absorption mode and switches into "Float transition, Linear Mode" which means 62.1V Right after switching to Float Transition BatteryAggregator takes the Min of the voltages, in this case the 60.75V from Akku1 and switches to it.

The expected Voltage is the one from the Akku2 which start to transition from the 62.1V to the float voltage of 60.75V. So the voltage from the Battery is slowly transitioning from 62.1V to 60.75V 
Currently it is directly changed from 62.1V to 60.75V

wetty, 2025-03-13:
Just as an idea. A second CVL mode
Cvl_floating: max_when_floating

When every Akkupack is in float or transition to float mode (mode string contains float), then use the max voltage of the packs.

So one mode for loading and one for floating.

Btw: I think that OVP does not exist in the floating process, as floating reduce the voltage and normally happens when balancing is done and no over voltage exists.

pulquero, 2025-03-15:
Please try v3.28 on main branch.

wetty, 2025-03-15:
Installed and CVL changed to max_when_floating

ewinkley, 2025-03-15:
> Please try v3.28 on main branch.

Sorry but how would I install 3.28?  3.27 is still showing up as most recent in package manager.

ewinkley, 2025-03-15:
> > Please try v3.28 on main branch.
> 
> Sorry but how would I install 3.28? 3.27 is still showing up as most recent in package manager.

I figured it out.

wetty, 2025-03-16:
v3.28 Is working perfect with the new mode max_when_floating!

Thanks a lot for this fix :-)

commit 4b605f34 (2025-03-15): Fixes #93. Add max_when_floating cvlMode.
closed 2025-03-16 by pulquero