pulquero/BatteryAggregator #94: File "/data/BatteryAggregator/battery_service.py", line 27, in <module>

ewinkley, 2025-03-15:
System details (very simple system):
Hardware: CerboGX 3.60~39
Qty 2 100AH Renogy batteries
dbus-serialbattery v1.6.20250131

After upgrading to 3.60~39, with BatteryAggregator 3.27, I lost access to the batteryaggregator.  I installed main (3.28) and no changes.

root@einstein:~# cat /data/setupOptions/BatteryAggregator/config.json
{
  "excludedServices": [
       "com.victronenergy.battery.ttyS7"
],
  "primaryServices": [
       "com.victronenergy.battery.ttyUSB0__0x30",
       "com.victronenergy.battery.ttyUSB0__0x31"
],
  "capacity": 200
}

root@einstein:~# tail -F /var/log/BatteryAggregator/current | tai64nlocal
2025-03-15 18:35:46.040200500     from pydoc import locate
2025-03-15 18:35:46.040568500 ModuleNotFoundError: No module named 'pydoc'
2025-03-15 18:35:47.686234500 Traceback (most recent call last):
2025-03-15 18:35:47.686436500   File "/data/BatteryAggregator/battery_service.py", line 27, in <module>
2025-03-15 18:35:47.687527500     from pydoc import locate
2025-03-15 18:35:47.687888500 ModuleNotFoundError: No module named 'pydoc'
2025-03-15 18:35:52.901390500 Traceback (most recent call last):
2025-03-15 18:35:52.901631500   File "/data/BatteryAggregator/battery_service.py", line 27, in <module>
2025-03-15 18:35:52.902752500     from pydoc import locate
2025-03-15 18:35:52.903124500 ModuleNotFoundError: No module named 'pydoc'
2025-03-15 18:35:54.576773500 Traceback (most recent call last):
2025-03-15 18:35:54.576947500   File "/data/BatteryAggregator/battery_service.py", line 27, in <module>
2025-03-15 18:35:54.578088500     from pydoc import locate
2025-03-15 18:35:54.578527500 ModuleNotFoundError: No module named 'pydoc'
2025-03-15 18:35:56.163239500 Traceback (most recent call last):
2025-03-15 18:35:56.163401500   File "/data/BatteryAggregator/battery_service.py", line 27, in <module>
2025-03-15 18:35:56.164530500     from pydoc import locate
2025-03-15 18:35:56.164921500 ModuleNotFoundError: No module named 'pydoc'
2025-03-15 18:35:57.755427500 Traceback (most recent call last):
2025-03-15 18:35:57.755599500   File "/data/BatteryAggregator/battery_service.py", line 27, in <module>
2025-03-15 18:35:57.756939500     from pydoc import locate
2025-03-15 18:35:57.757472500 ModuleNotFoundError: No module named 'pydoc'
2025-03-15 18:35:59.393174500 Traceback (most recent call last):
2025-03-15 18:35:59.393357500   File "/data/BatteryAggregator/battery_service.py", line 27, in <module>
2025-03-15 18:35:59.394509500     from pydoc import locate
2025-03-15 18:35:59.396225500 ModuleNotFoundError: No module named 'pydoc'

pulquero, 2025-03-15:
Looks like problem with opkg. Try 'opkg update'.

ewinkley, 2025-03-15:
No change after below commands.  I had to run the set-feed.sh in order to get opkg command to function.

root@einstein:~# /opt/victronenergy/swupdate-scripts/set-feed.sh candidate
Switching swu feed to candidate
resizing /dev/mmcblk1p3
resize2fs 1.47.0 (5-Feb-2023)
The filesystem is already 1310720 (1k) blocks long.  Nothing to do!

Switching opkg feed to candidate

root@einstein:~# opkg update
Downloading https://updates.victronenergy.com/feeds/venus/candidate/packages/scarthgap/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/candidate/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/candidate/packages/scarthgap/einstein/Packages.gz.
Updated source 'einstein'.

root@einstein:~# bash /data/BatteryAggregator/check_config.sh
This script will install required packages to use check_config.

The following packages will be installed:
  - python3-pip
  - jsonschema

Do you want to continue? [y/N]
y
Update packages...
Downloading https://updates.victronenergy.com/feeds/venus/candidate/packages/scarthgap/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/candidate/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/candidate/packages/scarthgap/einstein/Packages.gz.
Updated source 'einstein'.

Install python3-pip...
 * opkg_prepare_url_for_install: Couldn't find anything to satisfy 'python3-pip'.

Install jsonschema...
/data/BatteryAggregator/check_config.sh: line 32: pip3: command not found

Traceback (most recent call last):
  File "/data/BatteryAggregator/check_config.py", line 2, in <module>
    import jsonschema
ModuleNotFoundError: No module named 'jsonschema'

pulquero, 2025-03-15:
You will have to look at the release notes for 3.60~39 to check for any changes around opkg/python. As this is an rc version of venus, I don't intend to spend any time investigating because it could all be fixed in ~40. I wouldnt worry about /data/BatteryAggregator/check_config.sh working, but whether battery aggregator service runs or not.

closed 2025-03-15 by pulquero

ewinkley, 2025-05-15:
After upgrading to 3.60~74, I was able to get it working again - I appreciate your help.

root@einstein:/data/BatteryAggregator# ./setup
--- starting setup script v3.33 action: NONE
patching QtQuick 1.1 to QtQuick 2 in all .qml replacements

Battery Aggregator

Available actions:
  Install and activate (i)
  Reinstall (r) based on options provided at last install
  Uninstall (u) and restores all files to stock
  Quit (q) without further action
  Display log (l) outputs the last 100 lines of the log

Choose an action from the list above: i
++ Installing Battery Aggregator service
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/all/Packages.gz.
 * opkg_download_backend: Failed to download https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/all/Packages.gz, wget returned 8.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz.
 * opkg_download_backend: Failed to download https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz, wget returned 8.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/einstein/Packages.gz.
 * opkg_download_backend: Failed to download https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/einstein/Packages.gz, wget returned 8.
 * opkg_prepare_url_for_install: Couldn't find anything to satisfy 'python3-pip'.
++ Install Python library velib_python (master) from https://github.com/pulquero/velib_python into /data/BatteryAggregator/ext
--2025-05-15 22:01:57--  https://github.com/pulquero/velib_python/archive/refs/heads/master.zip
Resolving github.com... 140.82.112.4
Connecting to github.com|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/pulquero/velib_python/zip/refs/heads/master [following]
--2025-05-15 22:01:58--  https://codeload.github.com/pulquero/velib_python/zip/refs/heads/master
Resolving codeload.github.com... 140.82.112.9
Connecting to codeload.github.com|140.82.112.9|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘/tmp/velib_python.zip’

/tmp/velib_python.zip                              [     <=>                                                                                           ]  46.69K  20.4KB/s    in 2.3s

2025-05-15 22:02:01 (20.4 KB/s) - ‘/tmp/velib_python.zip’ saved [47815]

installing BatteryAggregator service
waiting for BatteryAggregator service to start
service BatteryAggregator running
complete - no errors

root@einstein:/data/BatteryAggregator# opkg update
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/all/Packages.gz.
 * opkg_download_backend: Failed to download https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/all/Packages.gz, wget returned 8.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz.
 * opkg_download_backend: Failed to download https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/cortexa7hf-neon-vfpv4/Packages.gz, wget returned 8.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/einstein/Packages.gz.
 * opkg_download_backend: Failed to download https://updates.victronenergy.com/feeds/venus/release/packages/scarthgap/einstein/Packages.gz, wget returned 8.

root@einstein:/data/BatteryAggregator# /opt/victronenergy/swupdate-scripts/set-feed.sh candidate
Switching swu feed to candidate
resizing /dev/mmcblk1p2
resize2fs 1.47.0 (5-Feb-2023)
The filesystem is already 1310720 (1k) blocks long.  Nothing to do!

Switching opkg feed to candidate

root@einstein:/data/BatteryAggregator# bash check_config.sh
This script will install required packages to use check_config.

The following packages will be installed:
  - python3-pip
  - jsonschema

Do you want to continue? [y/N]
[... 36 lines cut ...]
Configuring python3-setuptools.
Configuring python3-xmlrpc.
Configuring python3-image.
Configuring python3-pip.

Install jsonschema...
ERROR: Could not find a version that satisfies the requirement jsonschema==4.17.3 (from versions: none)
ERROR: No matching distribution found for jsonschema==4.17.3

Traceback (most recent call last):
  File "/data/BatteryAggregator/check_config.py", line 2, in <module>
    import jsonschema
ModuleNotFoundError: No module named 'jsonschema'

root@einstein:/data/BatteryAggregator# tail -F /var/log/BatteryAggregator/current | tai64nlocal
2025-05-15 22:07:12.844095500 INFO:com.victronenergy.battery.aggregator:Individual CCLs: {'com.victronenergy.battery.ttyUSB0__0x30': 50.0, 'com.victronenergy.battery.ttyUSB0__0x31': 50.0}
2025-05-15 22:07:12.844542500 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB0__0x30', 'com.victronenergy.battery.ttyUSB0__0x31']
2025-05-15 22:07:12.845658500 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.999899865820199, 'capacity'), (2.000100144207659, 'capacity')]
2025-05-15 22:07:12.857496500 INFO:com.victronenergy.battery.aggregator:CCL estimates: [99.99499329100995, 100.00500721038294], DVCC: True, user limit: -1
[... cut ...]
2025-05-15 22:07:12.888613500 INFO:com.victronenergy.battery.aggregator:Connected batteries: ['com.victronenergy.battery.ttyUSB0__0x30', 'com.victronenergy.battery.ttyUSB0__0x31']
2025-05-15 22:07:12.888618500 INFO:com.victronenergy.battery.aggregator:Current ratios: [(1.999899865820199, 'capacity'), (2.000100144207659, 'capacity')]
2025-05-15 22:07:12.888623500 INFO:com.victronenergy.battery.aggregator:DCL estimates: [99.99499329100995, 100.00500721038294]
2025-05-15 22:07:12.954450500 INFO:root:registered ourselves on D-Bus as com.victronenergy.battery.aggregator
2025-05-15 22:07:12.969591500 INFO:main:Registered Battery Aggregator com.victronenergy.battery.aggregator