pulquero/BatteryAggregator #96: BatteryAggregator not finding Batteries

Albarge, 2025-03-26:
I just transfered my system over from my Cerbo to Ekrano, BatteryAggregator V3.28 ran initially then after a reboot went missing from devices.

Ekrano V3.54 + Dbus-SerialBattery (No GuiMods)

I've tried uninstalling, reinstalling, stopping, restarting, rebooting.

[image]

my config: 

`{
  "primaryServices": {
    "com.victronenergy.battery.ttyS3": [
      "/SOC"
    ]
  },
  "currentRatioMethod": "capacity",
  "cvlMode": "max_when_balancing",
  "logLevel": "INFO"
}`

Any thoughts?

pulquero, 2025-03-26:
Check /SOC is correct in your config  usually it should be /Soc.

Albarge, 2025-03-26:
That worked! 

The command to check the config file didn't show it was a problem I guess it's not case sensitive? But handy to have, although It also didn't like "max_when_floating" so I went back to default.

closed 2025-03-26 by Albarge

pulquero, 2025-03-26:
"max_when_floating" added to config tool.