pulquero/BatteryAggregator #97: Possible to use a shunt for the Power calculation?

Albarge, 2025-03-26:
I have JK BMS's which don't have a very accurate internal shunt, so I use a SmartShunt for SOC + Current.

I tested using /Power from the shunt, but it just seemed to double up the Power of Aggregator + Shunt rather than only use the shunt.

{
  "primaryServices": {
    "com.victronenergy.battery.ttyS3": [
      "/Dc/0/Current",
      "/Soc",
      "/Dc/0/Power"

Is it possible to do this another way?

pulquero, 2025-03-26:
I'm not sure why power should behave any differently. Try voltage to see what happens.

Albarge, 2025-03-26:
I'll have to wait until tomorrow when there's some more decent charging loads to properly see variances against the  slower Victron averaging.

Albarge, 2025-03-27:
Well It is working now. after a few different tests and reboots. This is my working Config:

`{
  "primaryServices": {
    "com.victronenergy.battery.ttyS3": [
      "/Dc/0/Current",
      "/Soc",
      "/Dc/0/Power"
    ]
  },
  "capacity": 1020,
  "currentRatioMethod": "capacity",
  "cvlMode": "max_when_floating",
  "logLevel": "INFO"
}`

I'm not 100% sure what was happening, I think I tried adding a function in Dbus-SerialBattery after testing the Aggregator config settings and the GX data acting weird, but can't exactly remember and now it's working anyway..

I tried this:

`EXTERNAL_CURRENT_SENSOR_DBUS_DEVICE = com.victronenergy.battery.ttyS3
EXTERNAL_CURRENT_SENSOR_DBUS_PATH = /Dc/0/Current`

Adding the Shunt is possible in the latest version, but this populates all aggregated battery fields with the same shunt /Current data, and the power is derived from that, so it should not be used when aggregating. When I removed it and rebooted with the latest Aggregator config it finally displayed as I was hoping.

[image]

Although the config checker command is not running and I have some warnings about my BMS on USB3, but it looks like the data is says is missing is actually there when I look at the battery?

pulquero, 2025-03-29:
I don't think you can use the shunt both in EXTERNAL_CURRENT_SENSOR and primaryServices. I think I need to see some screen shots of dbus-spy to understand the issue better.

closed 2025-05-14 by Albarge