pulquero/BatteryAggregator #99: Charge Current Limit not Limited

wetty, 2025-04-04:
Hardware: RPi
Setup: DVCC On
SVS on, 
SCS On or Off - no difference

The Charge Current Limit is calculated correct from BatteryAggregator (from serial Batttery). The Controlling BMS in DVCC is Battery Aggregator.

But the Charge Current is not limited to the calculated value from BatteryAggregator v3.29. See the attached Screenshots
Any idea?

[image]
[image]

pulquero, 2025-04-04:
If DVCC is on, then the aggregated CCL is limited by the max current limit set in the UI (com.victronenergy.settings/Settings/SystemSetup/MaxChargeCurrent in dbus).

wetty, 2025-04-04:
You mean DVCC in general on?
[https://github.com/pulquero/BatteryAggregator/issues/65](https://github.com/pulquero/BatteryAggregator/issues/65)

Limit Charge Current does not change anything

pulquero, 2025-04-05:
Dvcc on means com.victronenergy.system/Control/Dvcc is set to 1

#65 is based on /Info/MaxChargeCurrent not the dvcc charge limit. 

cvlMode has to be set to "dvcc" to use the dvcc voltage limit.

wetty, 2025-04-05:
Actual the real charge current (15 and 21A) is much more than the charge current limit, shown in the BatteryAggregator parameter from 2.688A
I thought the current will limit to the value from BatteryAggregator

pulquero, 2025-04-05:
I need to see the log really.

wetty, 2025-04-06:
Loglevel is debug, but 
cat /var/log/BatteryAggregator/current
is attached. It contains several logs

[2025-04-06 current log.txt](https://github.com/user-attachments/files/19621192/2025-04-06.current.log.txt)

BatteryAggregator CCL is 16.35A
Overall Current for both battery Packs is 42,75A

[image]
[image]
[image]
[image]
 

Is there any other log I can provide?

pulquero, 2025-04-06:
Looking at the "more dramatic" log, I see
```
@4000000067f26e9b181ca544 INFO:com.victronenergy.battery.aggregator:CCL estimates: [1.28, 1.4679828178644964e+61]

So the aggregated current should be no more than 1.28A. I'm unclear where the 12.780A is coming from. What is your actual json config?

wetty, 2025-04-07:
This is my config
{
  "logLevel": "DEBUG",
  "currentRatioMethod": "ir",
  "cvlMode": "max_when_floating"
}

wetty, 2025-04-08:
I updatet to v3.30
Setting DVCC Maximum charge Current also sets the Charge Current Limit in BatteryAggregator. See the screenshots.
I have to check whether this changes when the current limitations kick in.
But is this correct that the CCL in BatteryAggregator is the same as the Limit in the DVCC Settings?
I thought BatteryAggregator is calculating the Limit from the CCL of the Battery Packs. Which are currently at 80A
(I have to admit I’m not sure whether the limit is only set in one place. And so shows always the minimum from DVCC setting and Batteries) 

[image]
[image]
[image]

pulquero, 2025-04-08:
From the log we have
```
@4000000067f51c5d0976998c INFO:com.victronenergy.battery.aggregator:CCL estimates: [32.064, 28.608], DVCC: True, user limit: 36
```

So the aggregated CCL should be 28.608A. The result is always clipped by the DVCC limit. If you have 36A in total that would imply the CCL estimates were greater than 36A at that point in time and got clipped.

wetty, 2025-04-09:
Thats correct.
But I wonder about the values in the log. Timestamp is 14:53:49 German Time
The screenshoot was taken 14:54:10 German Time. Just 21 seconds after the log entry.

From that the values in the log entry from my understanding should be more like 
CCL estimates: [13.73, 10.00], DVCC: True, user limit: 36 

Nevertheless BatteryAggregator Sets the CCL to 20A but this has no affect, as the battery is still loading with 36A 
(The last screenshoot shows all the necessary values)

Will see to have another example where the user Limit is not close to the charge current.

wetty, 2025-04-09:
Just to be clear
The calculation of the values ist not the Problem

The problem is that the CCL that is shown in the Parameters of BatteryAggregator has not affect.

See my very first screen shots.

pulquero, 2025-04-09:
Under "system setup", what's your "battery monitor" set to?

wetty, 2025-04-09:
[image]

pulquero, 2025-04-09:
Try rebooting the charger, sometimes you need to reboot it after a dvcc change. I would have thought the issue has to be with the charger not using the correct battery monitoring information.

wetty, 2025-04-09:
Problem with that is, that I constantly change the DVCC current limit via a node red flow. With that I set the limit to the amount the solarcharger produce. So only the solarcharger is loading via DC my batteries (97% efficiency) and the Multiplus II (85-95% efficiency) does nothing. All the other things i tried to get only DC from the MPPT did not work.

May i reboot everything when the batteries are nearly full and switch of my flow, to verify your information.

wetty, 2025-04-24:
Seems that a reboot after my last comment fixes this problem. Values now calculated correct.

[image]