pulquero/BatteryAggregator #101: VRM Shows 0% SOC after search for new Firmware in VRM

Newti04, 2025-04-16:
So, I am now a step further.
I take some more tests to find out the reason for 0% SOC in VRM.
It is only in combination with dbus-serialbattery and BatteryAggregator, when you check for firmware Updates in VRM. During the firmware check, VenusOS stops all Running serial connections to be able to connect to the MPPT-devices (VE.Bus). In that time all entries about the batteries are deleted on dbus and BatteryAggregator Shows 0% SOC.
Is it possible to take the last SOC instead of 0%, if the battery device vanished for a short time?

pulquero, 2025-04-18:
How about the voltage? is that not an issue too?

pulquero, 2025-04-20:
Fixed in main. Add "reconnectTimeout": 5 to the config to have a leniency of 5secs before a battery is considered as disconnected.

Newti04, 2025-04-24:
thanks for the new config option.
Because I had no config, I made a new config in /data/setupOptions/BatteryAggregator/config.json with
{
  "capacity": 840,
  "reconnectTimeout": 180
}
and tested it.
Some seconds after pressing "Firmware-Aktualisierung" in VRM, BatteryAggregator (v3.32) shows still the 0% SOC.
In dbus-spy the entries for dbus-serialbattery and the MPPT's comes back, when I leave the 'Firmware-Aktualisierung' menu. That is the reason, why I set reconnectTimeout to 180 seconds.
With my setting 4x MPPT and 3 serial-battery Firmware-Aktualisierung takes minimum 45 Seconds, till I can stop/end it.

I created the config.json with 1 entry for capacity with the index.html and added the one line for the reconnectTimeout.
But during the check with
bash /data/BatteryAggregator/check_config.sh I get this error-message:

`Traceback (most recent call last):
  File "/data/BatteryAggregator/check_config.py", line 14, in validate_json
    validate(instance=json_data, schema=schema)
  File "/usr/lib/python3.8/site-packages/jsonschema/validators.py", line 1121, in validate
    raise error
jsonschema.exceptions.ValidationError: Additional properties are not allowed ('reconnectTimeout' was unexpected)

Failed validating 'additionalProperties' in schema:
    {'additionalProperties': False,
     'properties': {'auxiliaryServices': {'anyOf': [{'items': {'type': 'string'},
                                                     'type': 'array'},
                                                    {'additionalProperties': {'items': {'type': 'string'},
                                                                              'type': 'array'},
                                                     'type': 'object'}]},
                    'capacity': {'type': 'integer'},
                    'classes': {'type': 'object'},
                    'currentRatioMethod': {'enum': ['ir',
                                                    'capacity',
                                                    'count'],
                                           'type': 'string'},
                    'cvlMode': {'enum': ['max_when_balancing',
                                         'min_when_balancing',
                                         'max_always',
                                         'max_when_floating',
                                         'dvcc'],
                                'type': 'string'},
                    'excludedServices': {'items': {'type': 'string'},
                                         'type': 'array'},
                    'logLevel': {'enum': ['NOTSET',
                                          'DEBUG',
                                          'INFO',
                                          'WARNING',
                                          'ERROR',
                                          'CRITICAL'],
                                 'type': 'string'},
                    'primaryServices': {'anyOf': [{'items': {'type': 'string'},
                                                   'type': 'array'},
                                                  {'additionalProperties': {'items': {'type': 'string'},
                                                                            'type': 'array'},
                                                   'type': 'object'}]},
                    'serviceName': {'type': 'string'},
                    'virtualBatteries': {'additionalProperties': {'anyOf': [{'items': {'type': 'string'},
                                                                             'type': 'array'},
                                                                            {'additionalProperties': {'items': {'type': 'string'},
                                                                                                      'type': 'array'},
                                                                             'type': 'object'}]},
                                         'type': 'object'}},
     'type': 'object'}

On instance:
    {'capacity': 840, 'reconnectTimeout': 180}

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/data/BatteryAggregator/check_config.py", line 55, in <module>
    is_valid, is_warning, message = check_json_file(config_file)
  File "/data/BatteryAggregator/check_config.py", line 45, in check_json_file
    return validate_json(data)
  File "/data/BatteryAggregator/check_config.py", line 22, in validate_json
    error_message += err.path[0]
IndexError: deque index out of range
`
With my setting 4x MPPT and 3 serial-battery Firmware-Aktualisierung takes minimum 45 Seconds, till I can stop/end it.

pulquero, 2025-04-24:
Check you have version 3.33, and branch/tag is set to main in package manager.

Newti04, 2025-04-24:
Thanks, after changing to main branch, I get the v3.33 and this works fine.

Just another short hint. 
If you add at the start of /data/BatteryAggregator/check_config.sh
cd /data/BatteryAggregator
then you can run the check also form /data/setupOptions/BatteryAggregator/ or other directories.

commit 113071ca (2025-04-24): Fixes #101. Use absolute path to schema file.
closed 2025-04-24 by pulquero