pulquero/BatteryAggregator #102: Could config file use a symlink for USB port mapping?

rldoose, 2025-04-22:
I tried creating a symlink for my battery monitor:
lrwxrwxrwx    1 root     root             7 Jan  1  1970 /dev/BMV712 -> ttyUSB5

I created /data/setupOptions/BatteryAggregator/config.json as follows:
{
        "excludedServices": ["com.victronenergy.battery.BMV712"]
}

It does not appear to be working.  Should it?  Could it?

Kind Regards,
Reagan

pulquero, 2025-04-22:
Why are you creating a symlink???

rldoose, 2025-04-22:
So that if the port assignment changes for the shunt, it does not break
what aggregator  ignores.   I’m doing 3 batteries plus 3  Victron devices
all via usb on a Raspberry Pi, so I have to use a hub.   Once in a while
the port assigned will change.

Sent from iPhone

pulquero, 2025-04-22:
Got you covered, see https://github.com/pulquero/BatteryAggregator?tab=readme-ov-file#service-names
Custom names can be set in the UI.

rldoose, 2025-04-25:
Awesome, worked like a charm!

Thanks