pulquero/BatteryAggregator #104: BatteryAggregator reporting double the current/amps

ewinkley, 2025-05-08:
System details (very simple system):
Hardware: CerboGX 3.55
Qty 2 100AH Renogy batteries
dbus-serialbattery v1.6.20250131
BatteryAggregator v3.33

Amps are reporting as what appears to be doubled for the batteryaggregator.  dbus-spy System/Batteries includes both physical batteries as well as the shunt, even though I thought I was excluding this shunt in the config.json.

**config.json:**
{
  "excludedServices": ["com.victronenergy.battery.ttyS7"]

  "primaryServices": ["com.victronenergy.battery.ttyUSB0__0x30","com.victronenergy.battery.ttyUSB0__0x31": ["/DC/0/Current"],
  "capacity": 200
}
config.json (END)

**dbus_spy:**
Services
com.victronenergy.adc
com.victronenergy.battery.aggregator                                                                                         00 Battery Aggregator
com.victronenergy.battery.ttyS7                                                                                                      00 SmartShunt
com.victronenergy.battery.ttyUSB0__0x30                                                                                               00 Battery 1
com.victronenergy.battery.ttyUSB0__0x31                                                                                               00 Battery 2
com.victronenergy.ble
com.victronenergy.digitalinputs
com.victronenergy.fronius
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.modbustcp
com.victronenergy.packageManager                                                                                                   Package Manager
com.victronenergy.platform                                                                                                                   Venus
com.victronenergy.rvc.can0
com.victronenergy.settings
com.victronenergy.solarcharger.ttyS6                                                                                             SmartSolar 100/50
com.victronenergy.system
com.victronenergy.tailscaleGX                                                                                                  TailscaleGX-control
com.victronenergy.temperature.adc_builtin0_8                                                                                          Storage Temp
com.victronenergy.temperature.ruuvi_db7a5abe8277                                                                                     Refridgerator
com.victronenergy.temprelay

**drill down on system/batteries:**

com.victronenergy.battery.aggregator
Alarms/CellImbalance                                                                                                                                            0
Alarms/HighCellVoltage                                                                                                                                          0
Alarms/HighChargeTemperature                                                                                                                                    0
Alarms/HighDischargeCurrent                                                                                                                                     0
Alarms/HighTemperature                                                                                                                                          0
Alarms/HighVoltage                                                                                                                                              0
Alarms/LowCellVoltage                                                                                                                                           0
Alarms/LowChargeTemperature                                                                                                                                     0
Alarms/LowSoc                                                                                                                                                   0
Alarms/LowTemperature                                                                                                                                           0
Alarms/LowVoltage                                                                                                                                               0
Balancing                                                                                                                                                       0
Capacity                                                                                                                                                   100.88
Connected                                                                                                                                                       1
ConsumedAmphours                                                                                                                                         -200.456
CustomName                                                                                                                                  00 Battery Aggregator
Dc/0/Current                                                                                                                                                21.35
Dc/0/Power                                                                                                                                                 285.01
Dc/0/Temperature                                                                                                                                               25
Dc/0/Voltage                                                                                                                                              13.3333
DeviceInstance                                                                                                                                               1024
FirmwareVersion                                                                                                                                                 0
HardwareVersion                                                                                                                                                 0
Info/BatteryLowVoltage                                                                                                                                       11.6
Info/ChargeMode                                                                                                                                              Bulk
Info/MaxChargeCurrent                                                                                                                                         150
Info/MaxChargeVoltage                                                                                                                                        13.8
Info/MaxDischargeCurrent                                                                                                                                      150
InstalledCapacity                                                                                                                                         199.736
Io/AllowToBalance                                                                                                                                               0
Io/AllowToCharge                                                                                                                                                1
Io/AllowToDischarge                                                                                                                                             1
LogLevel                                                                                                                                                     INFO
Mgmt/Connection                                                                                                                                              dbus
Mgmt/ProcessName                                                                                                       /data/BatteryAggregator/battery_service.py
Mgmt/ProcessVersion                                                                                                                                        v3.33
ProductId                                                                                                                                                       0
[... cut ...]
System/MaxCellTemperature                                                                                                                                      25
System/MaxCellVoltage                                                                                                                                         3.3
System/MaxTemperatureCellId                                                                                                                                Temp 2

[image]

Thank you for your assistance.

pulquero, 2025-05-08:
Remove primaryServices from your config, that is an Override config, batteries are picked up by default. The only thing that might make sense to do is this https://github.com/pulquero/BatteryAggregator?tab=readme-ov-file#aggregate-all-available-batteries-with-soc-info-provided-by-a-shunt-taking-precedence

ewinkley, 2025-05-08:
Thank you - confirming this resolved the issue.

closed 2025-05-08 by ewinkley