pulquero/BatteryAggregator #106: BatteryAggregator not showing on v3.60~72 and high CPU load

Albarge, 2025-05-14:
I have an Ekrano, testing v3.60~72, only on Gui-v1 at the moment, DVCC is on, I have had high CPU issues and trying to figure out the cause so uninstalled BatteryAggregator for a few days, I think generally v3.6x and Gui v2 are problems, but after installing BatteryAggregator today and rebooting, it didn't show up in settings anywhere, but it was using a lot of CPU:

[image]

I got repeated lines of this:

cat /var/log/BatteryAggregator/current | tai64nlocal

2025-05-14 16:24:48.098366500     from pydoc import locate
2025-05-14 16:24:48.098676500 ModuleNotFoundError: No module named 'pydoc'
2025-05-14 16:24:49.678997500 Traceback (most recent call last):
2025-05-14 16:24:49.679178500   File "/data/BatteryAggregator/battery_service.py", line 27, in <module>
2025-05-14 16:24:49.680124500     from pydoc import locate
2025-05-14 16:24:49.680453500 ModuleNotFoundError: No module named 'pydoc'

config:

`{
  "primaryServices": {
    "com.victronenergy.battery.ttyS3": [
      "/Dc/0/Current",
      "/Soc",
      "/Dc/0/Power"
    ]
  },
  "capacity": 1020,
  "currentRatioMethod": "capacity",
  "cvlMode": "max_when_balancing",
  "logLevel": "INFO"
}`

I'll try again on v3.55 but I was wondering if this is an issue for v3.60~72?

Albarge, 2025-05-14:
I'm not sure what happened, as I ended up having to do a full Factory Venus OS reinstall, everything is looking ok at the moment, so I'm sticking to v3.55 and Gui v1 for now to see how it goes.

pulquero, 2025-05-16:
Missing pydoc is an install issue. You just need to try the install again - there is an infrequent issue with libraries not always installing first time round (I think it is some sort of filesystem sync issue). You can install it manually 'opkg install python3-pydoc'.

closed 2025-05-29 by pulquero