pulquero/BatteryAggregator #107: DVCC Limit Charge Current vs. BatteryAggregator Charge Current Limit

wetty, 2025-05-28:
My System:
- Victron Multiplus II 5000
- Victron Solarcharger MPPT 150/45
- 2x JKBMS with LiFePo4 18s
- BatteryAggregator v3.33
- SerialBattery v2.0.20250324rc

In some cases BatteryAggregator is setting his Charge Current Limit to the value which is set under the DVCC Limit Charge Current.

Correct behaviour:
DVCC LCC = -1 (or off)
BA CCL = CCL Battery1 + CCL Battery2 (just to keep it simple)

This is also correct:
DVCC LCC = 0
BA CCL = CCL Battery1 + CCL Battery2 (just to keep it simple)
With DVCC = 0 the Victron Multiplus does not charge the Batteries, only the MPPT do charge! 
See the my discussion in the victron community:
[https://community.victronenergy.com/t/multiplus-ii-how-to-charge-only-with-mppt/29717](https://community.victronenergy.com/t/multiplus-ii-how-to-charge-only-with-mppt/29717)

All other values set in the DVCC LCC lead, in my opinion, to a wrong CCL Value in BatteryAggregator, as BatteryAggregator in this case just takes the same value as the DVCC Setting.
So setting DVCC = 15A leads to BA CCL = 15A.
The two values have to separate function. The BA CCL Value is to protect the batteries. So not to charge them with more than this value.
DVCC Value is the charge limit for the Multiplus. An additional Solarcharger is not affected from this limit! 
Example: 
- DVCC LCC = 15A, Solarcharger loading with 0A, Multiplus charges with 15A => Overall Charge Current 15A
- DVCC LCC = 15A, Solarcharger loading with 10A, Multiplus only charges with 5A => Overall Charge Current 15A
- DVCC LCC = 15A, Solarcharger loading with 20A, Multiplus only charges with 0A = Overall Charge Current 20A
 
As BatteryAggregator takes the same value as the DVCC settings, the Current is always limited to this value. Correct would be the same behaviour as for the DVCC LCC values -1 or 0. So always take the limits set in the BMS or even SerialBattery

Behavior ist the same for all the "currentRatioMethod" in BatteryAggregator

wetty, 2025-05-28:
May I haved quick and dirty fixed my issue.

I commented out the three Lines, so the user_limit ist always -1.

 dvcc_on = self._is_dvcc()
#        if dvcc_on:
#            user_limit = self.monitor.get_value("com.victronenergy.settings", "/Settings/SystemSetup/MaxChargeCurrent", -1)
#        else:
            user_limit = -1

        self.logger.info(f"CCL estimates: {cclPerBattery}, DVCC: {dvcc_on}, user limit: {user_limit}")
        # return 0 if disabled or None if not available
        if cclPerBattery:
            ccl = min(cclPerBattery)
            if user_limit > 0:
                ccl = min(ccl, user_limit)
        elif aggr_ccl.get_result() > 0:
            # CCL is available but no connected batteries
            ccl = 0
        else:
            # CCL is not available
            ccl = None

wetty, 2025-05-29:
The short answer: Yes that is an option.

The long answer: 
The DVCC Setting has nothing to do with the underlying BMS!
If you want to restrict the Charge Current in the BMS then use the BMS or an own BatteryAggregator Value.

With using the DVCC Value you change the behavior of this setting. Thats not a good approach.

What are the use cases to restrict the Charge Current on the BMS Layer to the value from the DVCC?
- If you just have one or more Multiplus as charger, then venus os limit the charge current to the DVCC value. So there is no need to control it on the BMS side again. 
- If you have Multiplus with Solarcharger then Multiplus + Solarcharger are charging together until the DVCC Value is reached. If the Solarcharger charge more then this DVCC limit, the Multiplus stops charging and only the solarcharger charge without limit.
This behavior makes sense as the Solarcharge are much more efficient as the Multiplus. With Setting DVCC to i.e. 15A the Multipus helps to charge until the Solarcharger delivers 15A ore more. Thats the best way to efficient charge your batteries and to take care that you charge enought to get them to 100%
To restrict the overall charge to 15A in this case does in my opinion makes no sense. Because then the power of the solarcharger is reduced, or converted from the Multiplus to AC. If you need AC Power then the ESS automatically tales Solarcharger power directly through the Multiplus.

Final question:
What is the behaviour of the
dvcc - bms is ignored setting?

Does DVCC than behave like the Victron Venus or the BatteryAggregator behaviour?

pulquero, 2025-05-29:
In a system with multiple solarchargers, no multiplus, charging a bank of AGMs with shunts then the DVCC value limits the overall max CCL (you can use virtual batteries to have individual battery CCLs). In this case, you want "bms+dvcc" mode. Hmmm, yeah, I should use the term "battery" not "bms".

> What is the behaviour of the
> dvcc - bms is ignored setting?

The battery aggregator ccl directly reflects the dvcc setting (this enables someone to directly drive the ccl value via dbus updates - niche, but someone is already doing this with cvl).

wetty, 2025-05-29:
Found in the Settings of dbus-aggregate-batteries from Dr-gigavolt

>  Please note: Victron ESS disables CCL (Charge Curent Limit) if DC-coupled PV feed-in is active.
> This program disables the DC-coupled PV feed-in when necessary to protect batteries
> (dynamic CVL - Charge Volgage Limit is activated) and re-enables again in safe condition.

With this I somehow understand why you take the DVCC Value for CCL.

pulquero, 2025-05-29:
To simplify config, I can scan for presence of any com.victronenergy.vedbus devices and auto-set cclMode to "battery" else "battery+dvcc". That should make the cclMode option transparent to the majority of users.