pulquero/BatteryAggregator #108: Instability with DVCC and Dual Daly BMS setup (charging/discharging oscillation & Float issue on alarm)

Micha854, 2025-06-13:
Hi,

I'm using a Victron setup without a MultiPlus, consisting of two identical DIY 300Ah batteries, each with a Daly BMS. Both BMS units are connected in parallel and individually wired via RS485 to the Raspberry Pi.

I'm using the serial battery driver from:
https://github.com/mr-manuel/venus-os_dbus-serialbattery
Version: 1.6.20250131

Both Daly BMS are correctly recognized and DVCC is enabled (as it always was before).

However, I’m experiencing two main issues:

1. Charging/Discharging Oscillation (with DVCC enabled)
As soon as the system voltage reaches around 27V, the system continuously switches between charging and discharging — it's flipping rapidly back and forth in a loop.
This happens only when DVCC is enabled.

2. Regulators drop to Float on BMS alarm (with DVCC disabled)
If I disable DVCC, the system behaves much more stable — basically as expected.
However, as soon as one Daly BMS raises an alarm, all charge controllers immediately go into Float, and the voltage drops to a steady 27V. This seems like an overreaction to a single BMS alarm.

I’m running the aggregator with default config. Here's my current config:
`{
  "primaryServices": {
    "com.victronenergy.battery.ttyUSB3": ["/Soc"]
  }
}`
This setup used to work just fine with a single Daly BMS.

Question:
How should I configure the aggregator (and DVCC?) properly to support a dual-Daly setup without these issues?
Is there anything I can do to improve the behavior — such as config tweaks, service prioritization, or ignoring certain alarms?

Which currentRatioMethod should I use?
Which cvlMode is appropriate?

Thanks in advance for your help!

pulquero, 2025-06-13:
A few clarifications:
Are you a new user or is this a result of upgrading to a recent version?
What is com.victronenergy.battery.ttyUSB3?
What is the reason to use DVCC?
If you don't have a multiplus how are you charging the batteries?
Thanks. Also do you have any logs from when it is misbehaving?

pulquero, 2025-06-13:
Service log, /var/log/BatteryAggregator/current.

Issue 2. Currently any individual alarm triggers an Aggregator level alarm. This is easy enough to change if we know which alarm and what behaviour we want.

pulquero, 2025-06-14:
Ccl looks ok to me, and shouldn't be affected by dvcc since you have no limit set. So that leaves cvl. To get cvl logging you need to set logging to debug (see the readme, you can do it in config, or live over dbus).

Is there a reason the serial driver is set to 3.45 and not 3.6?

pulquero, 2025-06-23:
No further response.

closed 2025-06-23 by pulquero