pulquero/BatteryAggregator #109: Installation wo setuphelper

endurance1968, 2025-07-09:
Hi
Is it possible to install BA wo setuphelper.
As guiv2 is now replacing v1 and at least I got several issues with 3.6x and v1 gui, installing/using setuphelper is more tricky. It does Not Look Like That it can/will support v2.
Cureently I went back to 3.55 but this can only be a Temp solution

pulquero, 2025-07-09:
Running
`/data/BatteryAggregator/setup`
should work.

pulquero, 2025-07-09:
It is the same difference. To install from scratch, download, unzip battery aggregator, download, unzip setuphelper. Run battery aggregator setup.

Currently, I don't intend to write my own independent setup script because well that is just duplicating code. I might consider bundling setuphelper scripts with battery aggregator so it's more seemless.